OpenAir-Lab/capstone-project #3: Research on Battery Management System

RawToast54, 2025-07-01:
Utilizing a USB-C to deliver 5V charging power to the battery. Research on the components needed to design a circuit to manage the distribution and safety measures for a lithium-ion battery.

ericmutton, 2025-07-02:
Here is the direct download for the [BOOSTXL-BATPAKMKII](https://software-dl.ti.com/msp430/msp430_public_sw/mcu/msp430/BOOSTXL-BATPAKMKII/latest/index_FDS.html) design files.

ericmutton, 2025-07-05:
This may also require evaluating whether to go single-cell or multi-cell.

Once the estimated power consumption of the system is known, current limits and system voltage settings can be defined in our specification.

Mobile amateur radio stations, such as those in consumer vehicles, are typically rated for 12A 13.8V supply power and output approximately 50W. In QRP applications, the ratings are lower, allowing for the use of 3.7V Li-Ion cells with lower current ratings (e.g. <10A)

RawToast54, 2025-07-19:
Current Progress on mapping and Organizing favored BMS

[image]

closed 2025-09-02 by AdrianneQuick