OpenAir-Lab/capstone-project #5: Select between ESP32-WROOM and ESP32-WROVER

AdrianneQuick, 2025-07-01:
All of my MCU research for this parent and sub issue will be located in the repository called [NOTION-merge](https://github.com/OpenAir-Lab/NOTION-merge.git) in the file [MCU-info](https://github.com/OpenAir-Lab/NOTION-merge/blob/7720a083adbd99a688af0d599e59c1ee48b8a8a7/mcu-info.md)

We need to choose between ESP32-WROOM and ESP32-WROVER. I will put my comparisons in the file I linked above. 

We also need to decide how much memory (flash) and PSRAM we should include with the board in order to select the right model.

AdrianneQuick, 2025-07-02:
Some questions that need to be answered:
- Will we be processing audio, doing any digital buffering, or storing modulated data streams?
- Will we have voice compression/decompression as a feature?
- Offline recordings of transmissions?
- Graphical displays & file system logs?
- Do we want to give target audience freedom to experiment? 
- Digital Signal Processing done with MCU?

ericmutton, 2025-07-04:
Having an SD card or PSRAM partition readied for settings persistence and messages (i.e. images, voice recordings) could address the offline transmissions aspect of this work.