OpenAir-Lab/capstone-project #21: Power Supply System

CURAC, 2025-07-19:
The power supply unit for this project is a smart charging and power management system. It is designed to safely and efficiently power the HAM radio from a universal USB-C SOURCE while managing the internal battery.

Its function entails three key operations:
  1. **Intelligent Negotiation**: Using a USB Power Delivery (PD) controller, the unit communicates with a USB-C charger to request a higher voltage (e.g., 12 V or 15 V) needed for the radio's circuits
  2. **Power Conversion & Charging**: It takes the negotiated high voltage and uses an efficient buck-boost converter to charge the internal Li-ion battery pack.
  3. **System Power Management**: It features a "Power Path" that allows the radio to be powered directly from the USB-C source, even if the battery is completely drained, ensuring uninterrupted operation while charging.

The following parts make up a part of the power supply. A sample BOM for visualization has been added below

[image]

ericmutton, 2025-07-22:
I restructured this power supply issue as a parent issue for all research tasks regarding power. 

@CURAC you mention the use of a different battery management system than the BQ25620 described in research task #3, the [BQ25792](https://www.ti.com/product/BQ25792). @RawToast54 what are your thoughts on this?

RawToast54, 2025-07-31:
USB-PD IC's

- [TPS25762-Q1](https://www.ti.com/lit/ds/symlink/tps25762-q1.pdf?ts=1753998654664&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FTPS25762-Q1)
- [TPS25751 ](https://www.ti.com/lit/ds/symlink/tps25751.pdf?ts=1753863250971&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FTPS25751%253FkeyMatch%253DTPS25751%2526tisearch%253Duniversal_search%2526usecase%253DGPN-ALT)

ericmutton, 2025-08-19:
@RawToast54 please let me know if you need assistance finishing the symbol and footprint import for your BMS recapture. Make sure to use the [system-testbed](https://github.com/OpenAir-Lab/capstone-project/tree/system-testbed) branch. https://github.com/OpenAir-Lab/capstone-project/commit/53f1a9e2cd4bad450e87ae31cc7b78fd570bf8c6