OpenAir-Lab/capstone-project #60: Design MCU PCB

AdrianneQuick, 2025-09-02:
Use Modulino format see #57

AdrianneQuick, 2025-09-04:
@ericmutton Would an 8bit one be too little?

ericmutton, 2025-09-04:
No, not at all, especially if we have on on each Modulino. For reference the keypad matrix being 4x4 needs a port (8 bits) exactly. Port 2 of the 16-bit one is so far the three DIO for the rotary encoder, so a 16-bit one is almost perfect for #33

ericmutton, 2025-09-04:
Heads up that the volume control may need to be on the #63 seeing as it is an analog signal. However, according to GROVE system compatibility #29 the I2C pins of a third Adafruit Stemma QT connector could be repurposed as Analog IO pins. Or we can do what [Arduino Knob](https://store-usa.arduino.cc/products/modulino-knob) board did but for a potentiometer.

closed 2025-10-23 by ericmutton